BeeHive-org/BeeHive #2: General Purpose Hub and related project questions

isobianin, 2019-11-21:
The analysis of the current state of the scientific research tools, which are designated for microbiology, shows absence of the modular platforms used for defined subject area. The FlyPi project (https://github.com/amchagas/Flypi) contains valuable assets for the design of the new PCB hub that is supposed to maximize given modularity by means of different connectors. More information is given in https://github.com/amchagas/Flypi/issues/34

The design of this hub is closely related to the design of the each discrete module and should be kept in mind at all times. These modules allow user to customize the hub for specific research. It should be possible to connect other modules to given hub. Nevertheless, new modules should also be viewed as the less expensive alternative to the ones on the market. 

As the main microcontroller, ESP32 was chosen which is bound to increase effectiveness of the FlyPi v2 in an every aspect.

This issue is intended for the discussion of the hub itself as well as the discussion of all other related questions that cannot be answered in other issues.

isobianin, 2019-11-26:
Hi @amchagas !
I have been busy with academic tasks and have not yet severely touched this project. I have thought a lot things through. On the other hand, I will have a very difficult time for the next month (the master’s degree defense). Just wanted to say that I work on this but this work is very slow due to the parallel education process. Probably, I will have a bit more free time two or three weeks from now.

isobianin, 2019-11-26:
How many modules will we have? This parameter corresponds with the power supply. Maybe the power supply from the previous version will be sufficient. In case it will not be sufficient, some enhancements will be needed

isobianin, 2019-11-27:
I will follow my question with some ideas. I think that the main hub PCB should have an array of connectors (up to 10 but it may vary) and power supply (PS), which should have enough power to deal with 10 different modules. The trick here is that if somebody wants to have only 2-4 modules powered up simultaneously, the powerful PS may not be cost effective. However, I think that such small power ratings are not that big of a problem and we can place PS for 10 modules without much cost added (it still will be added but not like 10 times I suppose). The Raspberry PS probably should stay, however, I do not know for sure. 
What do you think, Andre?

amchagas, 2019-12-01:
Hi @isobianin ,
What about having a modular PS as well? as in the system has a PS for the main board and that is able to "juice up" some devices, but if users decide to plug something power hungry, they could attach an extra power supply in between the main board and the new attachment?

isobianin, 2019-12-02:
Greetings, @amchagas !
Yeah, I see it and really like the idea. I will try to come up with some ways of how it can be implemented in practice

isobianin, 2019-12-10:
Just a quick update. 
Basically, I am now experiencing the peak of an academic workload. I had some unforseen tasks due to the fact that I had done a very good project. It is going to be sent to the national competition of the student projects and there is a lot of paper work I am in right now. I am estimating that this week or the next one I will finally have time to make progress on the flypi project

isobianin, 2019-12-22:
Hi @amchagas !
I have made some progress related to the general design and now we have some new information to discuss.

The current design of the main hub PCB is shown below (fig.1). Dimensions and layout are subject to change but they show the most likely outcome of the design.
[image]
Fig. 1 - The general layout of the hub

The main hub have in total 13 connectors with 4 pins in each one. These connectors are:
- 8 digital general-purpose connectors
- 3 analog connectors
- 2 I2C connectors

Additionally, there is another connector that will link 12v line with the 5v modular PS. The power consumption in this case should not exceed adequate boundaries because the main PS 12v5a can maintain up to 50W (approximately 80% of the maximum of 12V*5A=60W). The modular PS (fig. 2) will have two connectors, just in case. I thought that it may be used solely for the LEDs but one extra connector may come in handy. Just to clarify, you have mentioned that there should be dedicated PS for powerful LEDs and this is not it, of course.
[image]
Fig. 2 - The modular 5v PS

Digital connectors will power almost every module and therefore there are relatively a lot of them. Analog connectors are reserved for sensors, which may have analog signal as the output data. I2C ports are placed just in case they may be needed.
Also there is a "switch" in the layout (fig. 1). I have placed it there so the user is able to test and/or use the circuit with one LED as in the original FlyPi. All other output connectors depend on the circuitry they ecnounter in the driving PCB and they should not be connected directly to LEDs or other stuff.
Speaking about LEDs, I have spent some time on the idea of shift register based hub (fig. 3).
[image]
Fig. 3 - The low power LED hub

The TPIC6B595 is able to provide current up to 150 mA to each of its pins. So it is able to power some low consumption LEDs with no problem. But they are still power hungry and that is why PS (fig. 2) is needed more than ever. There are two variants of the PCB with 4 or 8 LEDs.
This hub is controlled via 2 connectors from the hub (->ESP32) due to the fact that each connector provides 4 pins but shift registers needs 5. Particularly, it needs 3 data signals and GND/5V. We can place 5 pin connector but I doubt that it will be used for other devices (fig. 4).
[image]
Fig. 4 - 4 or 5 pin connector

There are more things I started working on but they are not polished yet for discussion.
To sum up,
1) Any concerns or propositions that are related to the main hub design? (general layout, connectors, modular PS and transistor switch)
2) The LED hub, what you think about this concept? And what about 4 and 5 pin connectors?

amchagas, 2020-01-06:
Hi @isobianin!
Sorry for the delay, went away for holiday break... 
Overall I like your idea with the main hub, modular PS and transistor switches. I think this provides a good "agnostic" main hub, so that we/others can add different things as needed. 

Is there any particular reason you are focusing on LEDs and different ways of powering/interacting with different LED formats (sorry if I missed this). Don't get me wrong, I don't mind start with LEDs, just wanted to know if there was any particular reason for it...

On a related note, I stumbled on this project https://github.com/RoboticsBrno/RB3200-ESP32universal/ which has somewhat the same idea I guess?

isobianin, 2020-01-07:
Hi @amchagas, I hope you enjoyed your holiday!
Understood, I am glad that the general idea is right and the development phase may continue.

Such focus on the LEDs is based on the fact that this kind of module is relatively simple to develop (imho) and we have not determined whether there are some high priority objectives. Also, I see that LEDs are practically an integral part of the laboratory setup in one or another form (regular LED or UV). On a side note, I feel that it will be easier personally for me to make something simple and then move to more complicated stuff. Such modules as Peltier are already implemented in the previous version and estimated to be quickly turned into the compact PCB when needed. 

You have found a very useful piece of information! I see we can draw some inspiration from this repository.

isobianin, 2020-01-17:
Hello @amchagas! Here is, finally, the first major update. Firstly, there is a schematics of the project (fig.1)
[image]
Fig. 1  - The FlyPi hub schematics
It is made as readable as possible. There are 8 digital connectors, 3 analog, 2 I2C, 1 UART (has been added in the last iteration of the project) and a connector to the additional power supply. ESP32 is represented by ESP32-DevKitC V2 board. It is powered by 5V voltage that is supplied to the corresponding pin. 
The pinout is also present in the figure and it is GND - VCC - DATA1 - DATA2. Such type of connection is common in electronics, especially in personal computers. Furthermore, Grove sensors and modules can also be connected to the FlyPi hub without any additional connectors.  
The aforementioned schematics is turned into the PCB (fig. 2).
[image]
Fig. 2 - PCB layout and tracks
This is the current version of the tracing. As for now, it does not have any vias, only though hole pads. Width of tracks are taken from the first version of the FlyPi and the track for 12V line is the widest (1.5 mm). 5V line and GND are zones on the each side of the board.
[image]
Fig. 3 - Dimensions
Figure 3 illustrates dimensions, mainly the fact that they are the same as before. Previously, although I 
had estimated that PCB would be approximately 10 mm bigger in the each direction, the layout was minimized to fit in the enclosure. USB port and barrel jack were moved slightly due to the fact that their positions play minimal role in the enclosure design.

Right now, I am trying to organize repository and create BOM is a such way that it is similar to the original FlyPi and it can be processed by Kitspace.

isobianin, 2020-01-17:
As far as I understand, until we have more PCBs in this repositry, Kitspace recognizes this PCB and show its render and BOM. As of yet, I have not created a project there but all the necessary steps are completed so it is not a problem (done with BOM). We only need to discuss the latest update

isobianin, 2020-01-20:
Currently working on the LED/transistor switch hub and the power supply

amchagas, 2020-01-21:
Hi @isobianin This looks great! 
As it is, it already makes for a nice tinkering/prototyping platform I think. 
I have two questions: 
Seems the ESP32-DevKitC V2 is already on V4, is there any particular reason why you choose this board to start things? 
Also, from the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-devkitc.html) it seems that this board also comes in different flavours... do you know if all of them would have the same footprint and if their pin out is the same?

isobianin, 2020-01-21:
Hi @amchagas !

About the devkit board. I have written that it is V2 but in the BOM V4 is used due to the fact that it is their latest version. Sorry, got too carried away and forgot to change/mention this.

The pinout of the devkit board is the same between the versions (V2-V4). The choice of the esp32 option depends on the computational requirements. The final pinout of the board with the given esp32 is the same across all the options ([ordering information](https://www.espressif.com/sites/default/files/documentation/espressif_products_ordering_information_en.pdf) page 11). Just in case, I have checked esp modules again and their layout is all the same. The footprint of the devkit board does not depend on the modules it is carrying and is invariable.

amchagas, 2020-01-21:
great thanks!

isobianin, 2020-01-23:
Hi @amchagas !
Finished the power supply module and the LED/transistor switch hub for 4 elements.
PS schematics (fig. 1) is the same as it is on the FlyPi board and have two output connectors as discussed earlier.
[image]
Fig. 1 - PS schematics
Its PCB is shown below
[image]
Fig. 2 - PCB of PS 
LED/transistor switch hub schematics is also present (fig. 3) and completely corresponds with our depicted design.
[image]
Fig. 3 - LED/transistor switch hub schematics
PCB will look like this (fig. 4). 
[image]
Fig. 4 - PCB for LED/transistor switch hub
All files are in the main repository. BOM have all the same components from the FlyPi except for the [shift register](https://ru.mouser.com/ProductDetail/Texas-Instruments/TPIC6B595N?qs=sGAEpiMZZMtsbn1GaJysl1mvuihK4f5q5rvouWD%252BBjQ%3D). 

For now, I am waiting for the prototyping results. During this time, I will design the transistor switch hub for 8 output devices and will look into other modules we agreed to develop. And one more important thing. I have mentioned it in amchagas/Flypi#34 that we will need some kind of enclosure for all this stuff. I would like to design it too as a part of this project if there is a possibility to test it out. Dimensions that are used in PS and LED hub are already lined up with my idea for the future enclosure.

amchagas, 2020-01-24:
Hi @isobianin,
this looks great! Some minor comments:
On your power board pinout, input and outputs have a different pin number to track configuration, and this is not wrong, but I think it would make life easier, since we are thinking about having different add on boards, if we decided on a convention for it. In other words, would it be too much trouble to have pin1 be Vcc and pin2 ground on all the plugs?

I know this may sound like too much detail, but I've caught myself a couple of times trying to debug issued on a board, and having a pattern to what was connected to what would have made things simpler. This will also make it easier when making different cables for the different boards as we will know that pins1 and 2 will always be vcc and gnd no matter what the board being connected to it is (of course when vcc and gnd are going to a certain board). 

One extra thing is that if Vcc is pin1 the square layout of that pin will match the square layout of the +pole on the capacitor and on the collector of the transistor. 

To help things move forward, should I make an order for this boards and test them? we are currently developing a behavioural box for some experiments and this would be a nice test bed for it!

isobianin, 2020-01-24:
Hi @amchagas , I see what you mean. In the end, we are designing this system for the purpose of being convenient to the user, particularly us. So we change pinout format to VCC-GND-DATA2-DATA1 across all the modules and the hub itself, right? Nevertheless, it will not take long. 

And one small thing, I thought that the main hub is good by itself but it may be even better if we go for 4 layer PCB (for the extra good EMC). Just a suggestion.

Considering boards, I think it is a great idea! As soon as the latest changes are implemented, I will notify you.

isobianin, 2020-01-25:
Hi @amchagas ! It is done, changes are implemented through all the PCBs. Also, I set up links to  screenshots of the current design of PCBs and schematics in the "README.md" so it is easier to monitor updates (probably).

amchagas, 2020-01-25:
Hi!
nice, thanks for the fast update! Quick question, on the main board image there seems to be an unconnected capacitor on the top right?

isobianin, 2020-01-25:
It is connected to the copper pours but, yeah, I missed this. I will manually connect them so the whole supply circuit can work properly

isobianin, 2020-01-25:
Patched it up, now there is a proper track to connect PS to the copper pour

isobianin, 2020-01-27:
Hi @amchagas , added the switch hub for 8 connectors

amchagas, 2020-02-02:
hi @isobianin , just ordered the PCBs (hub, PS and 4 switch hub for now). There was a comment on the manufacturer website that things might take a little longer due to the Corona virus problem in China.

isobianin, 2020-02-02:
Hi @amchagas , I see. 

To keep things updated, I am now looking into the Peltier module and motor controllers and drivers

amchagas, 2020-06-04:
hi @isobianin! 
I started implementing one behavioural task using this project, right now this is as a fork here https://github.com/amchagas/BeeHive, but I'll make a pull request once there is something usable

At the moment I just managed to put out an explanation of the software stack I want to use and started coding a bit the ESP32 using micropython.

isobianin, 2020-06-06:
Hello @amchagas!
Okay, I got it. I am elated to hear that we will continue the development and it is great that this project may potentially gather more attention soon

isobianin, 2020-12-21:
The latest addition of "discussion" tab by GitHub renders this issue closed. Moreover, it has not been used for a long time and all new issues are quickly categorized in their respective "issues" sections

closed 2020-12-21 by isobianin