BeeHive-org/BeeHive #3: Name change suggestion

amchagas, 2020-02-17:
Hi @isobianin !
Since we are moving towards a flexible system that could be used in different types of experiments and settings, I would like to suggest a name change. Also this would bring the benefit of getting this device to create its own identity.

how about "lab swiss knife" or "openlabware lego"? (not great suggestions for the moment)

isobianin, 2020-02-17:
Hi @amchagas !
Sounds great, fully support your idea. I think that your suggestions are applicable. We can look into it. I need a little bit of time to come up with something too.

Moreover, further enhancing the scope of this issue, I would like to suggest that we can even design the logo. Not like something super fancy but I have an experience of working on my electronics department logo back in the university. Maybe it will be a good addition to a nice sounding name? Seems like a new task to complete.

P.s. Is the "lego" word copyrighted? Wow, it is [(link)](https://blockstad.net/things-you-should-know-before-using-the-word-lego/).

isobianin, 2020-02-20:
Hello @amchagas !
I have some propositions to discuss:
1. The "BeeHive"  biolab tool (not sure about the last part with the "biolab tool"). The meaning behind this name is that bees have their designated roles. This part corresponds with an idea of specific purpose modules. These bees are all doing their part to make the hive thriving. Personally, I think the last part relates to the "open source" ideology, particularly when we are trying to design something good and share it so we all may profit. The fact that the beehive is an artificially made bee colony also represents an implicit idea that our flexible system is made by people (meaning us) and for a purpose ("completing bioresearch tasks" is similar to "extracting honey", I guess). On another note, there are a lot of labs that are named "BeeHiveLab". Particularly, "beehivelab" exists here on github. I do not think that it is a major problem but it sure need to be considered. Last but not least, its phonetics is very nice due to the way it is pronounced. I may be overthinking this but I have considered this to be an important factor too. I am voting for this one.
2. "Symbiolab". It is derived from the word "symbiosis" to reflect the fact that all modules are co-exist together and make life for each other better. Pretty simple and straightforward. The word "symbiolab" is widely used for different research groups. It sounds nice but describes our system in a more vague way, meaning that "-lab" is just a broader term (imho).

Also, I have considered to name it simply "Apis" (Latin for the "bee") but it may be confused with the API (Application Programming Interface). Or "ApisHive" but the pronunciation starts to be more complex and strange.
There was an idea to change bees with ants but it did not work out for me. The meaning behind the beehive sounds much better.
Additionally, I thought about the hermit crabs that scavenge mollusc shells. There is a resemblance but it just does not sound phonetically right. 
I have also considered birds and marine lifeforms but they, as I think, are out of scope of this project subject field and will be irrelevant.

Andre, what do you think about all of these and do you have any additional suggestions?

isobianin, 2020-03-25:
Hello @amchagas !
I made 4 variants of the logo, supposing we had chosen "BeeHive" as the name. Quick description: 1 is the first variant made. It has an alternitive coloring scheme in 3 and also has some minor changes in stinger's sizing. Variants 2 and 4 depict my initial idea of a logo. The only thing I am not sure about is their dimensions. I mean, I think about adding one of these logos **on the PCB**. 5 is the same as 1 and 2 but with wings angled.

I was told the the **4-th** variant is the best so far.

1. [image]

2. [image]

3. [image]

4.  [image]

5. [image]

What is your opinion? Any suggestions?

amchagas, 2020-04-03:
Hi @isobianin, Sorry again for the late reply!

I like the name and the rational, as well as the logos! If you like the 4th one better, I would say stick to it :)

I'm going to try and update things on the hardware side, and do some testing. 

I would also like to point us to this project https://github.com/labthings

Very briefly, the idea is to develop a common framework/communication protocol to automate/facilitate communication between lab instruments. I think it would be a good idea to try and see if this hardware system could run with that protocol. What do you think?

isobianin, 2020-04-06:
Hi @amchagas !

First of all, changed name to "BeeHive". Yeah, probably will go with the 4-th logo variant.

Speaking about LabThings - I like the idea. Moreover, I think it is a good looking software solution for our design of lab tools.

closed 2020-08-19 by amchagas