BeeHive-org/BeeHive #5: Naming channels

amchagas, 2020-06-15:
At the moment, we used an "regular" naming convention of starting the digital channels from 0 upwards (and the same for the analog channels).

This makes sense, but makes it a bit hard to program the board, as these numbers do not reflect the IO of the board when calling functions.
Also, given the flexibility of each pin in the ESP32 (a digital pin, can be aslo a PWM, pin or an SPI, or a pin for a serial connection), this complicates things a bit.

I think it would make things more straight forward to name directly as their IO number, so for instance D0 would be IO33. A nice solution IMHO would be to keep the current numbering system, but add the IO reference next to it... (so something like DO/IO33) this way we can see straight away which pins can be used as analog, or SPI, etc. but still have an easy time programming things

isobianin, 2020-06-25:
Hi @amchagas!
I looked into this issue. I am inclined to agree that we better name the channels as they are in the ESP32. In order to safe the previous "regular" naming convention, we can have, say, IO31 on the front and D0 on the back of the PCB (fig. 1). It is double sided PCB and there should be no significant increase in cost. 

[image]
Fig. 1 - Combined naming on two layers

We may aswell try to stuff everything in a one text row but I fear that it may be unintelligible or it may not be printed at all (fig. 2)

[image]
Fig. 2 - Combined naming on one layer

If those variants do not meet expectations, we can always revert back to just IO ESP indicators without the current system.

amchagas, 2020-06-27:
Hi @isobianin 

From the two options, I think the first one is best (one writing on the back side). 

At the same time, now that I think about this again, I wonder if this is absolutely necessary? Is this nomenclature something needed for the board tracing/organization? 

Don't get me wrong, I don't mind keeping, it, I'm just wondering if having both wouldn't generate space for confusion?

isobianin, 2020-06-27:
D0...D(N) is great when working with some preprogrammed pieces of electronics but here it is, as our experience shows, a bit inconvenient. 
I believe that in our case we may in fact use solely ESP identification (like IO33, IO31). And I agree that less is more, especially with all the text that is up there on the PCB.

amchagas, 2020-08-19:
I think this issue is resolved. Mind if I close it?

isobianin, 2020-08-19:
Thought about it too, no problem, this is solved

closed 2020-08-20 by amchagas