BeeHive-org/BeeHive #6: Update name on board and make ESP orientation explicit

amchagas, 2020-06-15:
Some cosmetic changes:
- Since we moved on from FlyPi V2, it would be nice to see this reflected on the board as well.
- Right now there is no information about the orientation of the ESP32 on the board. an easy fix would be to add some text saying "antenna side" and "usb side" or something like that....

isobianin, 2020-06-25:
Hi @amchagas !
I totally agree with this. And I also would like to put our new logo somewhere. **But** I saw something very unpleasant and we need to tackle it. 

Indeed, we need to have some kind of a text that suggests correct placement of ESP. And here is where the problem is. ESP32 is placed in a such way that its USB port is facing the wrong way (fig. 1). I thought that USB faces the other way so it is akin to the previous FlyPi design. But now it is what it is.

[image]
Fig. 1 - ESP32 orientation

The question here is simple, should we change the design of the layout and rotate ESP32 180 degree or leave it as it is?

amchagas, 2020-06-27:
hi, 
I thought that the orientation of the ESP was intentional! :)  as this way, the antenna of the system is a more "isolated" area, without signals that could possibly interfere, or be interfered with running below it. 

but I really don't know enough about wifi transmission and signal corruption to know if this is necessary when the board is "lifted up" from the PCB. 

If signal corruption is not an issue, rotating the board is a good idea, if not too much trouble

I also noticed that you might have created your own schematics and footprint for the ESP? I was just wondering if you knew about third party schematics/footprint repositories like this one https://www.snapeda.com/parts/ESP32-DEVKITC-32D/Espressif%20Systems/view-part/ This might be handy to facilitate derivatives? and I wonder if it would be useful to add it to BeeHive?

isobianin, 2020-06-27:
Regarding wi-fi electromagnetic compatibility (EMC) - it will be alright, there is nothing to worry. However, there may be changes in how ESP is connected to connectors. Of course, it is possible to rotate ESP and safe the current connections but it will create a mess in tracing and its EMC.

I will try to rotate it and come up with a new tracing. After that, we should look into it and decide whether it is worth it. I mean, we just need to see what the other option before discarding it.

Regarding footprints, I have just checked it and, in fact, I used this very file that you addressed. Indeed, we should add it.

isobianin, 2020-06-29:
Hi @amchagas !

Here is an update. First things first, there are major changes in the net list. Here is a figure of ESP32 connections (fig. 1).
[image]
Fig. 1 - New connections
And here is an old one (fig. 2).
[image]
Fig. 2 - Old connections

The new design features (fig. 3):
- names of pins are changed to "IO(N)" to ease programming;
- new logo is added;
- PCB is renamed to "BeeHive";
- orientation markers are added;
- UART connector (the arrow is pointing to it) changed its position to this location;
[image]
Fig. 3 - The redesign 

I hope it sums everything up. I will be waiting for further suggestions.

isobianin, 2020-06-30:
A table to structure changes and make reprogramming a bit easier.

Table 1 - The new pin mapping
| Connector pin | Old | New | Note |
| ------------- | ------------- | ------------- | ------------- |
| D1-3 | IO33 | IO15  | |
| D1-4 | IO32 | IO2  | |
| D2-3 | IO26 | IO0  | |
| D2-4 | IO25 | IO4  | |
| D3-3 | IO14 | IO16  | |
| D3-4 | IO27 | IO17  | |
| D4-3 | IO13 | IO5  | |
| D4-4 | IO12 | IO18  | |
| D5-3 | IO10 | IO19  | |
| D5-4 | IO9 | IO21  | |
| D6-3 | IO11 | IO22 | |
| D6-4 | IO19 | IO23  | |
| D7-3 | IO8 | IO14 | |
| D7-4 | IO7 | IO27  | |
| D8-3 | IO15 | IO26 | |
| D8-4 | IO2 | IO25  | |
| A1-3 | IO35 | IO35  | Unchanged |
| A1-4 | IO34 | IO34  | Unchanged |
| A2-3 | IO39 | IO39  | Unchanged |
| A2-4 | IO36 | IO36  | Unchanged |
| A3-3 | IO0 | IO32  | |
| A3-4 | IO4 | IO33  | |
| I2C_1-3 | IO0 | IO9  | |
| I2C_1-4 | IO4 | IO10  | |
| I2C_2-3 | IO16 | IO12  | |
| I2C_2-4 | IO17 | IO13  | |
| UART-3 (TX) | IO1 | IO1  | Unchanged |
| UART-4 (RX) | IO3 | IO3  | Unchanged |

amchagas, 2020-06-30:
Hi @isobianin 
this looks great! thanks for the update. 
I'll work more on some code these days and upload it here when it is in a good state.

amchagas, 2020-06-30:
related to #7 we should think about removing the "CC" mark from the board and add either CERN OHL license or TAPR, depending on which one we like best...
here is a more detailed explanation on hardware licensing https://opensource.com/law/15/2/intro-open-hardware-licensing

closed 2020-08-20 by amchagas