BeeHive-org/BeeHive #7: adding license for the hardware itself and the documentation

amchagas, 2020-06-15:
Hardware projects need more than one license. 
Specifically, they need:
- a license for the software (Which is MIT)
- a license for documentation and designs
- a license for the hardware itself once it gets built. 

This is a bit of a long topic to put on the issue, and it is not super pressing at this stage... So I would suggest for us to take a look at [this material](https://openhardware4.me/open-hardware-leaders.github.io/Program/17_Week10.html#licensing) and circle back with suggestions for licenses soon.

isobianin, 2020-07-09:
Hi @amchagas !
Regarding CERN or TAPR choice, it was hard to spot differences . In addition to the [article](https://opensource.com/law/15/2/intro-open-hardware-licensing) you have mentioned, I read [rationale](https://ohwr.org/project/cernohl/wikis/uploads/0be6f561d2b4a686c5765c74be32daf9/CERN_OHL_rationale.pdf) of CERN and looked up TAPR. There is an another good comparison [article](https://helda.helsinki.fi//bitstream/handle/10138/312235/ki0119833enn.pdf?sequence=1) and in the end I think that CERN is the way to go. But do we need to register BeeHive as CERN-W or CERN-S? I think CERN-W as it has more licensing margin for software used.

amchagas, 2020-07-12:
Agreed!

closed 2023-02-23 by amchagas