BeeHive-org/BeeHive #10: Question about the 8 and 4 switch arrays NPN transistor connection

amchagas, 2020-08-12:
Hi @isobianin !

I was looking at the designs for the 8 and 4 switch arrays and I had a question (sorry I don't know if we cleared this before):

[image]

On the connections for the NPN transistor, shouldn't the transistor itself be connected to the ground line of the header pin, instead of the +5V?

In more detail shouldn't this be: +5V going to J3 1 and J3 2 being connected to Q1 1 and Q1 3 connected to ground?

isobianin, 2020-08-12:
Hi @amchagas !

No worries, we did not talk this through. This switch is done exactly as it was designed in FlyPi (Pic. 1)
[image]
Pic. 1 - NPN switch design
I thought that this design works in FlyPi and that the circuitry has already been programmed and tested. Thus, it was left as it is. If this type of switch needs redesign, we can implement your idea or think about something else.

amchagas, 2020-08-12:
oh, that was def. a design flaw on my side. It didn't lead to critical issues on the FlyPi design, but I correct it on a new update of the board I made (porting it to KiCad 5 mostly). I would correct that here as well if you don't mind... 

And sorry for the mistake that got propagated here!

best,
Andre

isobianin, 2020-08-12:
Got it. To be honest, I too let this discrepancy find its way into the new design. I will make the changes and post an update.

isobianin, 2020-08-16:
Hi @amchagas !
I have been redesigning PCB for the new layout with 74HC595 and noticed something in a circuitry. Mainly, I have connected transistors this way (fig. 1):
[image]
Fig. 1 - Common emitter
And you suggested to connect them this way (fig. 2) (maybe I have misunderstood your directions):
[image]
Fig. 2 - Common collector (?)
So do we use common emitter or common collector?
I believe that if we want to make a simple switch, we should refer to the first design with common emitter. I am not sure whether the second variant will work due to the fact that emitter in NPN-transistor is doped higher than collector. As alternative to the both schematics, we can use a voltage follower.

amchagas, 2020-08-16:
Hey, 

Yes you are right.  Figure 1  is the way to go. Sorry for the confusion!

closed 2020-08-19 by amchagas