BeeHive-org/BeeHive #11: Question about power connector in 8 and 4 switch arrays

amchagas, 2020-08-12:
Hi @isobianin !
Another question related to these boards (sorry if it has been covered already!):

[image]

I was wondering if J1 1 and J2 1 pins should really be left unconnected? I see that the VCC for the shift register is coming from J11 but in this way the system is "locked" in 5V, I mean the shift register can only take up to 5.5V but we might want to use the transistor switches to switch, say 12V. so if the IC VCC comes from  J11, we can't use 12V. On the other hand if VCC to the IC comes from J1 and J2, then we use J11 to feed other voltages to the switch arrays?

Let me know what you think!
best,
Andre

isobianin, 2020-08-12:
I see what an idea is. It would be a great addition. Although, I am yet to think of an answer how to do it. I mean, if J1-1 and J2-1 are used as 5V, that means that 5V power is coming from ESP32 pins and I doubt that ESP will supply enough power, especially when the breakout is used for things other than lighting.

Actually, there are few things here I want to discuss. First of all, these switches were designed specifically for 5V (as you have mentioned, IC cannot pass more current/voltage than stated in datasheet). Secondly, it is envisioned that for high-power LEDs, a new board with completely separate power supply (12V/5A or more powerful) will be designed.

amchagas, 2020-08-12:
I don't know if I got this wrong, but I understand this system in this way:

The shift register just sends 5V out to the base of the transistors (so it is not high power), adding to that, the 5V coming from J1-1 and J2-1 are not coming from the ESP itself, but rather from the 5V regulator (using one of the LM2596) built in the main board (so there should be enough power for the shift register?).  

If this indeed works this way, then J11 is "free" for us to use with more flexibility, either 5V again from the LM2596 line or 12V from the 12V line. Since the transistors will be the only "connection point" between J11 and J1-1, J2-1 and this point will be the ground of both circuits, it should be fine to connect it this way?

isobianin, 2020-08-13:
Regarding J1-1/J2-1, you are right, my apologies.

Regarding 12V line, I see what you mean. If we are to take into account a new layout, it is certainly possible. Nothing restricts us from connecting 12V to the board (J11). So, in order to get things right, I will summarise:
- J1-1 or J2-1 will be used to power IC*
- J11 will be used to power transistors with 5V or 12V
*note: we could potentially power IC with J1-1 and transistors with J2-1 but when J11 is simultaneously connected, it will result in one of power supplies getting burned.   

This way, I will add dedicated power track for J1-1 or J2-1 to IC and power polygon will now be able to supply 12V/5V.

amchagas, 2020-08-13:
Great, thanks!

amchagas, 2020-08-13:
Hi, 
I am re-reading the datasheet for the TPIC6595 and now I see what you mean with the power LED  design. Since the pins of the shift register low-side open drains, so they shouldn't be able to "switch on" and NPN transistor in the config. I described, no? 
And if I understand correctly, in this new proposed configuration, we should not need the NPN transistor? Or is there a major missunderstanding from my side?

isobianin, 2020-08-13:
We may use LEDs without transistor solely with TPIC. But, if my hunch is correct, those transistors ([note](https://forum.arduino.cc/index.php?topic=112498.0)) were used to drive LEDs in order to lower current needed from pins (in FlyPi case, it was arduino pins). Here each pin can provide up to 250mA. With 12V addition, it is possible to drive even more powerful LEDs.

isobianin, 2020-08-13:
New design uploaded with "ver 2" affix in "hardware" folder

isobianin, 2020-08-14:
Hi @amchagas !
Are we going to make further changes to the design?

amchagas, 2020-08-14:
hi @isobianin,
Thanks for creating a new version so fast! 
So I took a look at the note you sent, but that seems like a bad example? I mean the schematics there seem wrong? from there, they also describe a different shift register, namely the sn74hc595. 

Looking at its [datasheet](https://www.ti.com/lit/ds/symlink/sn74hc595.pdf), specifically at point 10.2, it looks like this one has a different configuration from the  TPIC6595. Looks to me that the sn74.. is a "high side" shift register and the TPIC6595 a "low side" one? 

What I mean is that on the TPIC, setting any port of the device to "high" will simply open the internal gate and allow current to flow to ground. On the Sn74, turning any ports to high actually sends a voltage out of the device (as shown in the datasheet, point 10.2) that could actually be used to turn on an led, or forward bias another transistor.

If this is the case, then the current design for our 4 or 8 transistor switch is possibily not working as intended?

I don't know if I am getting this all wrong, sorry if I am, it is just that I never worked with this shift registers!

isobianin, 2020-08-14:
I see what you mean. Indeed, you are right and TPIC will not efficiently work that way. I am sorry that it turned out this way. With this in mind, it seems that we can design switches this way:
- replace TPIC with sn74hc595;
- use TPIC without transistors;
- use TPIC with transistors and think of a way of their implementation;
To be honest, when I had looked at the new transistors' placement, I immediately thought about going with 74HC595. My fault not seeing this before. How about we consider this option first? Are there any other ways we can improve on this design?

amchagas, 2020-08-14:
No worries! 
I think using the sn74hc595 is a good solution, it will allow us to be more flexible in terms of the transistors that can be used later on. I mean, if we need to change the transistors later, or if anyone else wants to derive this model into something else, they will also have an easier time changing it.

isobianin, 2020-08-18:
Hi @amchagas !
Both switches design is updated

amchagas, 2020-08-19:
Great, thanks! I'll look into making the boards and testing them etc..

isobianin, 2020-12-11:
Hi @amchagas !

Can we consider this issue closed? To summarise, here we implemented the following:
- making it possible to use 12V or 5V
- changed shift registers to the current ones

I believe that they were tested a long time ago, the issue is solved

amchagas, 2020-12-17:
indeed!

closed 2020-12-17 by amchagas