BeeHive-org/BeeHive #13: Question about new modules

isobianin, 2020-08-18:
A gas sensor and temperature/humidity sensors are added. However, it seems that their price may be higher than those modules that are already available on a market. Despite this, I still believe that our platform should have its own modules solely to create a self-reliant system which can be used without external hardware but in the same time it permits use of other sensors and actuators. Is this principle correct?

amchagas, 2020-08-19:
That is a good question :) I agree that we should have some working examples for sensors that would allow people to start measuring things easily. Would you mind sharing the links to the ones you mention are already available in the market?

isobianin, 2020-08-19:
Gas sensor - [this one](https://www.robotshop.com/uk/mq-135-gas-sensor.html?gclid=CjwKCAjwm_P5BRAhEiwAwRzSOzzDRR1FHIbQBditxaqoWS2_gvpAI1d5-0FjPeMK9pfALNct6j-_4hoCp-sQAvD_BwE) and [more](https://www.robotshop.com/uk/catalogsearch/result/?q=gas+sensor&order=relevance&dir=desc) on this website; [grove](https://wiki.seeedstudio.com/Grove-Gas_Sensor-MQ2/) and there is more
Humidity and temperature sensor - [grove](https://www.seeedstudio.com/Grove-Temperature-Humidity-Sensor-DHT11.html), [this](https://aliexpress.ru/item/32951977722.html) and counting. There is also DHT22 version, which is bigger in size and better in accuracy.

It is good to hear that we are doing the right thing with the development of additional modules.

This issue is related to #14

amchagas, 2020-08-20:
considering issue #14 maybe we should think about:
- designing additional modules that are not present on the grove library?
- creating examples on how to use a few selected sensors that are in the grove library?

isobianin, 2020-08-20:
I think that would be the case here. #12 is definitely needed and I will focus on that. I will add more explanation in the respective issue as soon as I will have sufficient information

Regarding grove examples, yes, indeed it is a good solution.

isobianin, 2020-12-21:
This issue was answered a long time ago and all new modules that need clarification have their own "issues" sections. Furthermore, in one of the latest updates, I adjusted gas and temperature/humidity sensors to have the pinouts exactly like Grove's ones.

closed 2020-12-21 by isobianin