BeeHive-org/BeeHive #14: Making beehive GROVE compatible

amchagas, 2020-08-19:
Hi!

This is a bit of an annoying one but might be useful in the long run...

I was checking some boards at Seeed Studio for another board and realised that we the plugs we have are almost compatible with their Grove system, where they offer a lot of different sensors and actuators.

The main difference is that Grove uses a 4 pin system (same connector as ours I think), with 1 and 2 being "data" lines (digital or analog), 3 being VCC and 4 being GND 

This below is one example of a sound sensor plug schematic that only uses one data line. The sensor is this [one](https://wiki.seeedstudio.com/Grove-Sound_Sensor/)
[image]

I was wondering if it would make sense to make our system compatible, so that we gain access to all of these sensors/actuators "for free"? 

I know it will be a bit of a pain to re-order things, but maybe worth it in the long run? Also, I am happy to split the task so that we each take half the modules and update things.

What do you think?

amchagas, 2020-08-19:
actually [here](https://wiki.seeedstudio.com/Grove_System/) they have a whole description of the system with board sizes and pin mapping for digital, analog, i2c, etc

isobianin, 2020-08-19:
Hi @amchagas !

To be honest, it is a bit painful to see us coming back to this layout. Let me explain. On January 24 (check out our [general questions chat](https://github.com/isobianin/BeeHive/issues/2)) we decided to get away from GND VCC pinout because it was easier to debug. I apologize here for not backing up my decision but before 24 January pinout had been specifically made exactly the same with groove for the reasons you mentioned here (tested and cheap modules and there are a lot to choose from).

We may change the layout again, revert to what it had been before changes. On the other hand, can we just simply use wiring that has "free" headers-connectors? I mean, we can simply twist wires and connect "power" to "power" and "gnd" to "gnd" and data1 and data2 pins are the same. 

Generally speaking, I thought from the get-go that our system should be compatible with the closest analogous systems so I think it is wise to implement it, especially if we are remaining true to our open source policies. 

Regarding time, I think I can handle it and do it relatively quick because I spent lots of time recently redesigning switches and other sensors.

amchagas, 2020-08-20:
So sorry for this!! This is obviously my fault. For some reason I didn't realise that we were moving away from Grove!!! 

I can change the layouts, this way you don;t get the extra work for something that was on my end and you can focus on developing new circuits which are grove compatible? Just let me know, so that we are not both working on redesigning things

isobianin, 2020-08-21:
Hi @amchagas !

No worries, we will easily solve this discrepancy. Okay, I can focus on the other stuff. If there are any questions regarding PCBs, I am open for discussion

closed 2020-08-24 by amchagas