BeeHive-org/BeeHive #16: High power LED driver

isobianin, 2020-09-21:
From what I was able to find, Grove has two modules which can be used with powerful LEDs:

- https://www.seeedstudio.com/Grove-LED-Strip-Driver.html

- https://www.seeedstudio.com/Grove-EL-Driver.html

However, they are relatively powerful. There are other examples that, probably, more appropriate:

- https://www.aliexpress.com/i/32993594548.html

- https://www.pcboard.ca/3-watt-pwm-star-led-driver

A long time ago, we determined that a high power LED driver should be present within BeeHive system. The question is, if we are to do it, what are the specifications for it? 
- Mainly, what LED(s) is it going to drive? It is highly likely that this question is related to a question whether is it going to be powered by 12V5A power supply or something new. 
- Should we use plastic JST connectors or terminal blocks for wires? 
- What about heat dissipation of an LED. Can it be cooled via heatsink or there are some specific requirements? What if an LED driver requires heatsink too? From the cost-effectiveness point of view, I think that we should try to omit anything that wastes energy and/or requires additional care/maintenance.
- How many high power LEDs are needed in the system simultaneously?
- How will an LED function in the system (where does it go and how is it going to be mounted there) and does it require any mechanical enclosure or something similar? I presume that FlyPi's solutions are applicable here.

amchagas, 2020-09-23:
Hi!
I am surprised that there are no more options on the Grove ecosystem to drive LEDs. Thanks for researching on this and for finding other options as well. 

I think we have too much of a loose concept on what are "high power leds". From experience what researchers call high power, is not necessarily what engineers would call high power ;) For biological applications, I think up to 3W leds should be more than enough.

I do not have very precise answer to your questions, but here is some direction:

1- I would stick to designs that can be driven by the 12V 5A power supply. If we need more power later, I would stick with 12V, but then get a power supply that has more amps. 

2 - I would imagine the JST connectors should be fine, since they can take up to 3A (https://docs.rs-online.com/f25e/0900766b81357f2f.pdf), and I think we should be below that in most cases? 

3 - If we can manage with heatsinks/passive cooling only, that would be ideal.

4 - can't know for sure... Sorry. Maybe let's design something that takes one led per channel/digital signal control and rely on the shift registers in case we need many?

5 - the mechanical details on how these are going to be held can change a lot I think. 
I am thinking we can use them for the following: - fluorescence, optogenetics (both similar to flypi), but also to control "day/night cycles" in behavioural boxes and or incubators
I think there can be so much variability in here that spending time to design a holder now is not the best time investiment

Let me know if this helps, or if it only adds confusion? Maybe a simple opamp current driving design with the option for the user to change the output current would be a good place to start?

isobianin, 2020-09-23:
Thanks for the detailed answer. I got the gist. There are a lot of questions from my side because I want us to determine what characteristics should the module designed have. As far as I can tell, it is alright.

For the starting point, yeah, I will look up to opamp circuitry and some other variants. I will post updates as there will be an advance with the development.

isobianin, 2020-12-14:
Hi @amchagas !

I was looking into this issue lately. I think about developing the design based of the foundation of the peltier module. I looked up OpAmp circuitry that you had proposed but I see that it has some overcomplicated aspects for what is easily achieved by the said MOSFET drivers. Yet, we already saw the results of the MOSFET drivers and they are plausible.

Regarding shift registers, I think to make a simple circuitry with four switches for a start. To be honest, I don't think that we will even need more than that. The shift registers will be the same as in our previous BJT arrays. The circuitry will have 2 drivers for 4 MOSFETs and all the complementary components. That is how is it for now

isobianin, 2020-12-14:
Hi @amchagas , while I am on this topic there is one more thing. I think controlling these LEDs with PWM is quite effective. On the other hand, it is somewhat difficult task to achieve with shift registers as in this case we would need to constantly turn the snifter on and off with the given frequency.
A good question on this topic.
https://electronics.stackexchange.com/questions/110744/pwm-driven-led-array-over-a-shift-register
There are libraries for 74hc595 registers for Arduino that do exactly this. Will it work or should we just connect IO pins directly to the drivers? There are also timer ICs like 555. I feel that we need to decide what approach we should take before commencing further development (I will just design only those parts of schematics that I am adamant about)

amchagas, 2020-12-17:
hum, this is an interesting question. maybe as an straightforward approach should we just connect the driver directly to the IO pins? there should be enough for the number of LEDs that might be needed in biological applocations, and we can use the shift register for other devices?

isobianin, 2020-12-17:
Hi @amchagas ,

precisely what I was thinking and referring to. Simple yet functionally complete and easy to program

amchagas, 2020-12-20:
👍 Great!

closed 2021-05-28 by isobianin