BeeHive-org/BeeHive #18: System power management issue

isobianin, 2020-10-17:
Work on the Peltier driver design showed that the BeeHive system has some flows in regard to its power management system that is currently implemented (fig. 1). To put it simply, there was a strange feeling that something is not right in a way all the cables are connecting and, sadly, it turned out to be true.

[image]
Fig. 1 - Current power management system

As seen in the figure, the system cannot work as intended. Although it can be modified "on-site" but it is not the way it is supposed to be.
 
To be precise, 12V5A line is simultaneously represented by 4 pin connector (main hub and 5V power supply) and 2 pin connector (newly developed Peltier driver and 12V5A breakout). It is a 100% design issue and must be tackled. This issue leads to another one where 4 and 8 switch arrays cannot be fed with 12V5A line from the main hub because the connectors are different.
Those issues can be dealt with in 3 ways:
1) the least amount of redesign - we change 4 pins to 2 pints only on hub and 5V3A PS. Actually, it can be done even with the current version where on a place of 4 pins we solder 2 pin connector
2) the moderate amount of redesign - we change the whole 12V5A line to 4 pins because it is better from a standpoint of a current management. Even so it has its own flaws discussed in #12, I believe that it is indeed a good idea that actually was partially implemented a long time ago (the main hub has 4 pins for 12V5A). **The pinout in this case is** + + GND GND. In the same time, every module shown will have some changes.
3) a safe ideal variant - change 12V5A to 4 pins but with a new connector like [this](https://www.google.com/url?sa=i&url=https%3A%2F%2Fmy.rs-online.com%2Fweb%2Fp%2Fpcb-headers%2F3922565%2F&psig=AOvVaw0NDKhZTodLV2axwVmPIEX2&ust=1603025021242000&source=images&cd=vfe&ved=0CA0QjhxqFwoTCPCnxdzTu-wCFQAAAAAdAAAAABAJ). It is the most difficult variant yet it will be the safest as 12V5A line will be distinctive from all other lines. To think of it, maybe this variant can be implemented as _a user experience feedback_.

Another issue to mention is that I, unfortunately, slipped up and the **Peltier driver** has the **reversed pinout** as well as 12V5A, temperature and gas sensors. I am already onto this.

I will redo this stuff but we just need to decide what is the best way of power management. In my opinion, we can do it this way: take the 1) option and then look for option 2) and 3) after the field test.

isobianin, 2020-10-17:
A little update, it turned out that temperature/humidity and gas sensors are all alright, they had an old silk layer with wrong names for pins but the pinout is traced correctly

amchagas, 2020-10-22:
Hi @isobianin, 

Thanks for raising these issues! I have ordered a couple of boards that have these footprints and connectors, so I would suggest that I put them to use, test and report. This way we can cover these issues and if more issues arise while I am testing things, we can tackle all of them at once?

Also this will be a nice pressure for me to get things going on my side with software and task implementation!

isobianin, 2020-10-22:
Hi @amchagas ,

I think tackling all issues at once will be a nice way to do it. Sorry that we have to "debug" all of these, especially some very simple mistakes, using the real prototypes in the first place.

isobianin, 2020-12-11:
Hi @amchagas !

Today, I realized a comprehensive update on this issue. Most of the modules were quite easy to update. I was also able to fix some small things along the line. The peltier module can now be equipped with heatsinks and it has two connectors for 12V5A power lines. I guess this issue can be closed after testing. More detailed update log is in the 'description' of the 'summary'.

amchagas, 2020-12-17:
Hi @isobianin !
Thanks for the update! 
I am going to start end of the year break, but will order the boards so that they are here once I get back to test things and get it going.

closed 2021-05-28 by isobianin