BeeHive-org/BeeHive #19: Shift register on 4 output signals boards

isobianin, 2020-12-17:
Questions have arisen while working on a high power LED switch array. It is highly likely that such array have a sufficient number of LED outputs which is 4. Therefore, the questions are:

1) Why bother with shift registers and addition of software libraries (referring to [this](https://github.com/isobianin/BeeHive/issues/16#issuecomment-744390693) comment), if an array board takes 4 pins from ESP32 regardless whether they operate a shift register or directly connected to drivers?
2) Could we omit a shift register on 4 switch array as it seems that it only adds unnecessary cost and complexity to the board? In FlyPi schematics, the given transistors worked fine while being directly controlled by Arduino pins.

amchagas, 2020-12-17:
interesting. Is your idea just having an 8 pin shift register and scrap the 4 pin one?

isobianin, 2020-12-17:
Hi @amchagas ,

not exactly. My idea is to get rid of a shift register (component) so the 4 switch array has no over-engineering. I believe that it is still an important asset as it will be relatively cheap and simple device while more complicated one (8 switch array) will have more outputs to offer

amchagas, 2020-12-20:
Hi!
Ah ok, I see. Indeed that makes a lot of sense. I would go for that (removing the shift register altogether).

isobianin, 2020-12-21:
Made these changes in the latest update and moved the previous version (ver 3) to the "archive"

closed 2020-12-21 by isobianin