BeeHive-org/BeeHive #20: Documenting the project

isobianin, 2021-01-24:
There are various potential routes to take in terms of making the project as open as possible. Here is the paraphrased information by Andre (amchagas) from the email chat we had the other day which can be implemented one by one in the following order:

1) using instruments provided by github such as its own wiki;
2) using git-page to create a welcome-page;
3) register the project on hackaday to get more people interested/involved;

Having spent some time looking into it, I do not know how to correctly parallel this process and think that if we design a simple github-wiki and then add a welcome page, it will be a good way start the whole documentation process as for now it is pretty much consists of some random notes without a centralized 'database'.

amchagas, 2021-01-26:
Hi Ihor, 
I think this approach is nice, the only thing I would watch out for is the possibility of exporting the documentation created on GH wiki. What I mean is: Is there a way to simply download all the documentation in case we want to port it to another platform and/or for instance edit it offline?

amchagas, 2021-01-26:
Nevermind! Just saw that one can clone the wiki locally. This solves my question