BeeHive-org/BeeHive #21: Write board versions on boards and documentation?

amchagas, 2021-01-26:
Since we are going through a few implementations of the boards, I think it would be useful to have this information printed on the boards, as this would help with knowing, when debugging and/or helping others implement the system, which board specifically we are discussing. 

This would not be a lot of work I think, but would help prevent difficulties that come from people chatting about different board versions without knowing...

isobianin, 2021-01-26:
Hi @amchagas ,
I completely agree on this one. However, I am not sure how to implement this straight away. Does it mean that each module will have its own version number? Or does it mean that the whole BeeHive system will have a version number which increments with each release? What number do we start with?

amchagas, 2021-01-26:
That is a good question. 
here is a document I like for versioning software https://semver.org/ maybe we can use that as a base? 

- Something like 0.1.X for the current version, where the boards can have X in different numbers? or maybe this could be just confusing.

- as we have all the boards as independent units (as in, I could have an arduino controlling the peltier board, if that was the only thing I had available), we could just number them independently? Then we could have a version number for BeeHive and clearly state somewhere that this version of BeeHive contains boards X,Y,Z in their respective versions 0.1,0.2 and 0.3?

isobianin, 2021-01-27:
Hi @amchagas ,

I would go for the second option where each board has its own version and the start-up page/wiki home page will explicitly tell what is the current version of BeeHive and what corresponding versions of boards are included in that release. 

I like the software versioning convention, should we mention it in the documentation?

isobianin, 2021-02-15:
Hi @amchagas !

I have updated design on 5V3A breakout and implemented the new numeration convention. What do you think? It is simple as v0.1 here (fig.1).

[image]
Fig. 1 - 5V3A PCB

If it is alright, I think I could add it on every board

amchagas, 2021-02-15:
Hi @isobianin ,

I think this looks good. Would add it to the other boards! Thanks!

isobianin, 2021-02-17:
Hi @amchagas , agreed

amchagas, 2021-04-14:
I think we can close this one?

isobianin, 2021-04-14:
Hi, @amchagas, indeed

closed 2021-04-14 by amchagas