BeeHive-org/BeeHive #22: upload boards to kitspace

amchagas, 2021-04-13:
Hi @isobianin !
As I am about to order components to solder some boards and send them out to you and Thomas, it would make it easier for me if we uploaded the boards to kitspace (so that I can order components with the 1-click bom extension).

Are you confortable with that? As in do we think the current boards are in a stable enough state to be uploaded?

isobianin, 2021-04-13:
Hi @amchagas !

I have been checking the boards and I think they are good to go, yet I would like to point out two things. First of all, we had some problems with the MOSFET switch. About 4 month ago I implemented changes to the board where I rearranged MOSFETs in way that they can be equipped with heatsinks. Actually, I changed almost the whole layout as I also implemented another connector for 12V5A line. The aforementioned board is still to be tested but I think it is going to be okay. Secondly, about 2-3 month ago I did a small revision of all boards. Apart from adding version number to each of them, I did some small changes like adding an LED to 5V3A power supply module, changed orientation of connectors to match Grove's ones, etc. Hope we will get good results from prototyping.

amchagas, 2021-04-14:
Hi,
thanks for the update on what was done to the boards!! 
I will test the ones that I added to kitspace and once we know both the kitspace synching and the boards work fine, I will add the remaining ones

closed 2021-04-14 by amchagas