BeeHive-org/BeeHive #23: create GH organization to host different example repositories

amchagas, 2021-04-13:
hi! @isobianin 
At the moment, I am working on a few projects that are incorporating beehive boards: 
- in line heater
- incubator
- behavioural system

they are a bit spread in different places and I think it would be useful for organisation purposes to create a Github organisation and have different repositories there. This way all example use cases have each their repository with bill of materials for the parts that are not the electronic boards, as well as use/build instructions etc. This will prevent the beehive "central" repository to become too big/confusing to manage...

This is as always a suggestion, and I would be happy to discuss pros/cons and different implementations..

isobianin, 2021-04-14:
Hi @amchagas ,

I looked into the possible other alternatives for the sake of discussion:
1) Hackaday
What I understand from hackaday is that it is great for sharing and simultaneously blog posting.
2) Google code or google drive
Though I classify them as alternatives, I ma not sure that they will be as "visible" from a search engines as github but I do believe it manageable to store multiple projects on either of them

I think going with github organisation is a good idea. Additionally, looking through the platforms above got me thinking whether we may need a blogging style of an account somewhere (hackaday, twitter, etc)?

amchagas, 2021-04-14:
Hi @isobianin 
I think Hackaday is a great platform as an add-on to what we are doing on github. some features are not present on Hackaday,io (such as version control). 

since we have already a lot of things placed here, I would stick to this space and use other platforms as you mention, sort of as an blog update system. This would help with getting the word out about the project and hopefully get others interested!
Here is an example of another org I created to keep another project organised https://github.com/Open-2-Photon-Microscope myabe it gives a better idea?

About Twitter, I think it is a great platform if you are thinking about getting the word out about your work and projects. this is at least my experience with some of my projects. It will require a bit of time to post things and keep it updated etc... But on the other hand, you get to learn about some great tools/projects from other people...

isobianin, 2021-04-14:
Hi @amchagas ,

I too think that sticking to github is a good idea. Besides, I really like the implementation of "organisation". I thought it might get messy but is really neat and well structured. Though, I wonder, do we need to make each module (peltier module, sensors, main hub) as an independent repository? 

Regarding blogging, I think we may try Hackaday and Twitter when we have results and, most importantly, photos of devices.

amchagas, 2021-04-14:
I think for now I would leave this repository as it is, since it is aggregating all the boards and documentation about the electronics 
We can add repos that show users how to use a couple of the boards (and link it to this repo and to the specific boards needed), together with other parts to make devices. 

Later if we feel this arrangement does not work, we can change it. 

Alternatively, GH has a submodule feature, where generally speaking a folder in a repository can have another complete repository inside it. This is probably better explained here https://gist.github.com/gitaarik/8735255

But since this adds complexity, I would for now, just create an organisation and create new repositories as needed, while keeping the central one as is.

isobianin, 2021-04-14:
Alright, I agree with you on going with the organisation option. What is the next step? Who is allowed to create organisation? Should I try create it? Is there a specific algorithm (creating organisation, making you an official collaborator/owner with all rights, etc)?

amchagas, 2021-04-14:
if you have  a moment, yes please create it. Once it is done you can add people with different permission levels and also rebase repositories to the org if you want to have this one hosted directly there..

isobianin, 2021-04-14:
I created an organisation and sent you an invitation. A link is https://github.com/BeeHive-org/BeeHive . But I see that the repository I cloned there, well, just a clone, so it does not follow updates from this repository. Is that right? Does it mean that we will have to manually pull updates from this repository to the organisation's one or vice verse?

amchagas, 2021-04-14:
hey, yes, so if you just cloned the repository there, we would need to pull changes when we want updates to be brought in. 
We could solve this by deciding to only work on one repo and then bring changes to the other one... this might become a bit problematic when more people start collaborating with this project... 

Another solution would be to go into the settings of this repository and change where it is based. 
If you go to settings and go all the way to the bottom, you will reach the "danger zone" there you can find:

```
Transfer ownership
Transfer this repository to another user or to an organization where you have the ability to create repositories.

```

Since you are also the owner of the beehive org, this should be ok?

isobianin, 2021-04-14:
Hi, I see it. I think it might work

amchagas, 2021-04-15:
I think we are done with this one as well.. Closing!

closed 2021-04-15 by amchagas