BeeHive-org/BeeHive #30: peltier board questions/suggestions

amchagas, 2021-05-14:
I am currently working on a behavioural setup for another project and using BeeHive as the main electronics. Here are some suggestions based on what little experience I gathered so far:

---

change the NC pin on one of the input connectors to VCC -> in this project I am using the peltier boards  to drive two solenoid valves and one linear actuator. In none of those applications I have a temperature sensor, but I still need to connect the second input plug to get 5V into the board and power certain ICs in the board. If both input pins have VCC, for this application the second input plug ns not needed, and there are less cables, parts being used....

---

Related to the point above, maybe it would be useful to remove the temperature circuitry completely from the peltier board. and create a dedicated board where multiple temperature sensors could be connected (since we can address them individually diigitally). This would make the peltioer board one function (right now there are two, temp sense and h-bridge) - we could then also rename the board...

isobianin, 2021-05-14:
Hi @amchagas ,

I see what you mean. Okay, I will implement the changes. However, I have a question regarding the H-bridge/peltier driver. If we are going to make it general purpose, meaning that it will be used for all kinds of electronics (just as you mentioned, it may be peltier, solenoids, motors?), maybe I could look into its protection circuitry? We have talked about that behind the scenes but just to elaborate on that, internal diodes of MOSFETs are enough to sustain reverse currents from solenoid and other inductive components, provided they are low power. If this circuitry is going to be used for motors then I would like to check whether implementation of additional protection in form of diodes is needed. What do you think?

amchagas, 2021-05-15:
Hi Ihor, 
I think looking into protection mechanisms is good, I would however make sure what the diodes in the MOSFETs are capable of handling, maybe this is already enough for our use cases... Maybe even for motors they are enough?

isobianin, 2021-05-17:
Hi @amchagas !

I looked into MOSFETs and they should be enough for our application. In essence, there is quite a lot of information how to do a proper protection of H-Bridges but there is little to no information on how to determine whether the internal/intrinsic/body/parasitic diode of MOSFET is suitable for the job. I found two main points that reinforces my believe that our schematics will work fine:
1) internal diodes usually have almost the same characteristics as a MOSFET itself. In our case, our MOSFETs are well above the specified modes of operation;
2) the body diode is less effective in high-frequency applications due to its characteristics. I believe that we will just have to field-test whether motors we will be using work fine here. "Less effective" means that they discharge reverse current slower than a discrete one. Therefore, if motors are used to drive something mostly in one direction, we will be fine;
3) again, due to the fact that we have quite a margin on our MOSFETs, thermal regime should be alright too.

Regarding your other request, could you please look into the following figure (red arrow on fig. 1) and answer whether this is what you want to see.
[image]
Fig. 1 Pins of driver
It is not that hard to provide 5V to **all** NC pins but I am not sure whether this is needed. Could you please tell me, where 5V/VCC voltage is more convenient for your applications (what pins should be changed to 5V)?

amchagas, 2021-05-18:
Hi, good to hear about the MOSFETs, and that we should be on the safe side with them. 

Regarding your question, I would like to leave the plug that deals with the temperature sensor as it was (so NOT adding the trace you have in the figure) and add a VCC line on the other plug so it would be [GND, VCC, IN2, IN1] isntead of [GND,NC,IN2,IN1]. This way if I am not using a temperature sensor, I can still power the rest of the ICs in the board, by just using the other plug and its connections... does that make sense?

amchagas, 2021-05-18:
ah, wait, now that I look closer at the board, the other ICs are powered by 12V directly?

isobianin, 2021-05-18:
Hi @amchagas ,

ICs we are using here are MOSFET drivers so they are powered by 12V line, you are correct. I see now where you were going with that VCC line.

I think I will still change pinout to the one you said: [GND, VCC, IN2, IN1]. This way, the trace will become shorter and there is no need in second plug. Yes, this is it, when the driver was being developed it needed the second connector for the sake of temperature sensor but now it may be excluded from the layout. Or am I missing something here?

amchagas, 2021-05-19:
No, you are not missing anything!
Thanks!

isobianin, 2021-05-19:
Okay, got it. I will implement the changes

isobianin, 2021-05-20:
Hi @amchagas,

could you please check out an updated layout for a rebranded [driver board](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/driver)?
[image]
Fig. 1 - Updated layout

amchagas, 2021-05-21:
I think this looks great! thanks for the update!

isobianin, 2021-05-21:
Hi, no problem. I think we can close this issue then

closed 2021-05-25 by isobianin