BeeHive-org/BeeHive #31: board suggestion: solenoid controller

amchagas, 2021-05-14:
For an [odour stimulator project](https://github.com/Sussex-Neuroscience/odour-stimulator), I created a board that delivers a "spike and hold" signal. This was based on the [schematic](https://www.theleeco.com/engineering/electrical-engineering/lee-solenoid-valve-drive-circuit-schematics/) offered by the company that makes the solenoid valves that we are using for that project. 

I would like to suggest to adapt the created board so that it works with BeeHive.
- Right now the board takes in 24V to drive  12V solenoid valve. If we want the BeeHive version of the board to have the same specs, we would need to add a step up regulator to the board, so that it would take the standard 12V from beehive, convert it to 24V for the spike period of the signal. 
- Other than that, we would just need to change the connectors in the current board, as the board, the way it is, already has inputs to drive 2 solenoids.

isobianin, 2021-05-15:
Hi @amchagas ,

I was quite puzzled with the statement "the board takes 24V to drive 12V solenoid". I looked into schematics that is provided by 
the manufacturer and that is present in your repository. Am I right in an assumption that solenoids in this system work the way described by the manufacturer in the first passage? (fig. 1)
[image]
Fig. 1
I think that this is it because otherwise it is quite hard to understand why we need to step up 12V to 24V to then step it down again to 12V.

Regarding the implementation, yes, I think your idea is right. I will look into how to implement step up regulator and change connectors

amchagas, 2021-05-15:
Hi Ihor, 
The figure/explanation you shared is why the board needs 24v to drive a 12v solenoid. From what I understand this could be applied to any solenoid, and not just the one from this specific manufacturer?

Therefore we could add this circuit to step up the voltage, and the odour stimulator then becomes beehive compatible/driveable.. (I wonder if there are any ICs that allow for either step up/down depending on a combination of resistors? - this would make it quite flexible, no?)

The other option would be to skip the step up and make sure we use lower rated solenoids.

Thanks for looking into this!

amchagas, 2021-05-15:
You can also see below how the signal driving the solenoid looks like.. 

[image]

isobianin, 2021-05-17:
Hi @amchagas ,

thank you for the information provided. I have been looking into this and I too think that the recommendations from the manufacturer could be applied to all solenoids, regardless of their origin (except those that explicitly mention that their solenoids cannot be used under these conditions).
Regarding the step up regulator, I would like to ask quite a simple question: what is the best outcome for us? My opinion is that if we need 12V solenoids and "spike and hold" signal then we will try to reach this goal rather than changing it to omit implementing drastic changes.

amchagas, 2021-05-18:
Hi Ihor, 
the advantage that I see is that by using 12V solenoids we can add this odour stimulator to the "portfolio" of tools developed with BeeHive. Also as far as I know, a lot of solenoids use 12V by default... (but this is from observing, and not actually researching things).

isobianin, 2021-05-18:
Hi @amchagas ,
based on the information you provided, I think we should stick to 12V solenoids. I will look into how to make a step-up regulator or what can be done with this issue. Right now, I am thinking about making a step up regulator as a discrete module and finding one that can be bought. What do you think? Are there any other things in the system that may require 24V (like motors, servos)? If not, than a step-up regulator on a PCB with solenoid driver seems as a rational idea

amchagas, 2021-05-19:
Hi Ihor, 
thanks for looking into this. It is hard to tell if we ever going to need 24V for other applications. I think we should aim for not needing that, but hard to say with 100% of confidence. 

At the same time, a step up as a discrete module is a good general solution, but I wonder how much current we would be able to get from the step-up circuit (as far as I understand in step up applications you end up with a higher voltage having to "sacrifice" current?) to make it useful for other applications? (I really do not know how much motors, servos that are running at those voltages would need in terms of current?). For the solenoids we are using, I know that they are fairly low current... 

Also, for the solenoid board, having an isolated step up would mean we take the 12V from the main board, step up to 24V (for the spike phase of the signal on the board) then step it down again to 12V (for the remaining of the signal) on the solenoid board, maybe that is a bit wasteful? But than again if it works, it should be fine, since having a generic step up solution might be best?

Another argument on the solenoid board, is that having an external step up would allow us to use the solenoid board with different types of valves... I mean, if we use a 6V rated valve, we do not need the external step up, and can simply use the step down on the board. this would make the signal be 12V in spike phase, than 6V for the rest of the signal (I hope this makes sense?)

I have no preference relating to this, and would go with what solution you think it is best!

isobianin, 2021-05-19:
Hi Andre,

thanks for sharing a detailed opinion on this! Truly, I cannot seem to find a middle ground here, though I am gravitating towards a discrete regulator more. Yes, regulators are all about converting one component of power, voltage or current, into another, but power will remain constant in every case (or rather, it can only change via decreasing).

isobianin, 2021-05-21:
Hi @amchagas ,

take a look at this [step-up converter](https://www.aliexpress.com/item/1005001266721719.html?spm=a2g0o.productlist.0.0.656d7118CGmXlN&algo_pvid=e9a4d3f1-05fe-4c2f-9c81-0a6182b63f83&algo_expid=e9a4d3f1-05fe-4c2f-9c81-0a6182b63f83-2&btsid=0b0a556a16215243280922697e41ae&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_). In essence, it is what we need: it is relatively cheap, adjustable and is proven to work.
Regarding our own implementation of something along these lines, it is still complicated. I spend time investigating possible solutions. I am still unsure what approach to take so that will take time. Regardless, I think the module that I mentioned above is worth checking out.

isobianin, 2021-05-21:
Hi @amchagas ,

take a look at [this step-up converter](https://www.aliexpress.com/item/1005001266721719.html?spm=a2g0o.productlist.0.0.656d7118CGmXlN&algo_pvid=e9a4d3f1-05fe-4c2f-9c81-0a6182b63f83&algo_expid=e9a4d3f1-05fe-4c2f-9c81-0a6182b63f83-2&btsid=0b0a556a16215243280922697e41ae&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_). In essence, it is what we need: it is relatively cheap, adjustable and is proven to work.
Regarding our own implementation of something along these lines, it is still complicated. I spent time investigating possible solutions, though I am still unsure what approach to take so that will take a while. Regardless, I think the module that I mentioned above is worth checking out.

amchagas, 2021-05-21:
Hi @isobianin , I think the module looks good.

maybe an approach would be to buy some of them, test them out with the system and make decisions from there? 
I think having some hands on experience with it could help more informed decision?

isobianin, 2021-05-21:
Hi @amchagas ,

well, it is certainly a good decision, provided there is a budget for it. Honestly, I think that they will come in handy in all sorts of projects

isobianin, 2021-05-26:
Hi @amchagas ,

I would appreciate if you were to take a look at [the solenoid board](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/spike_n_hold) and [a discrete boost converter](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/12_24_boost_converter). The only thing that I was not able to fully complete are BOMs. In the solenoid board, I did not change it at all and in a discrete converter I filled in all rows except for LM555. 
The boost converter looks like this (fig. 1):
[image]
Fig. 1 - A boost converter
The most important thing in it is that I tried to make booster from the components we already have and it worked. This is why both schematics have detailed descriptions how and which components can be changed.

Also, I almost forgot, solenoid board have a connector for 24V in case a user have a discrete converter (fig. 2). This way, there is no need to install other components. It can be additionally used to provide 24V as an _output_ from the solenoid driver but I do not know whether it can be used somehow.
[image]
Fig. 2 - 24V connector on the solenoid driver

isobianin, 2021-05-27:
Hi @amchagas ,

I have a question regarding coils that we are using. In the solenoid controller board you used coils that are much smaller than those that had been used in FlyPi board and later adopted by BeeHive. From both footprints I can deduce that they are not interchangeable. The question is, what inductor are we going to end up using? It would be convenient to have only one type of inductor across the system

amchagas, 2021-05-27:
Hi @isobianin,
the boards look good! and great idea implementing the boost with the parts we are already using!!

About the inductor, we should stick with the same model and footprint we are already using. The different one on the other solenoid board is likely an oversight from my side...

isobianin, 2021-05-28:
Hi Andre, 
splendid. I changed the footprint, should be fine now.

isobianin, 2021-06-02:
Hi @amchagas ,

I made some changes to the discrete module, mainly, I added another connector that can output 24V just in case there are multiple low amp 24V modules (fig. 1)
[image]
Fig. 1 - Two 24V output connectors

amchagas, 2021-06-28:
Boards are arriving this week for testing!

In the meantime, I was thinking about the input signal pins. 
Right now, they are 2X 2pin, with input and ground. To easy compatibility with the rest of the system, we could simply replace them for 1X 4 pins that is compatible with the connectors leaving the central hub.

I made the necessary changes and uploaded them as a pull request: https://github.com/BeeHive-org/BeeHive/pull/32

isobianin, 2021-06-28:
Hi @amchagas ,

have you looked into this thoroughly? I remember that we thought through a convention regarding high ampere signals, in particular, that where high amperes present, two connectors are used. On the other hand, not all applications require a lot of amperes, so we settled to have two connectors "just in case" but commonly use only one. If you are using boards right now and it is more comfortable to have 1X 4pins connectors then there should be a lot more changes across the whole system.
As an example, [the main board](https://github.com/BeeHive-org/BeeHive/blob/master/hardware/hub_PCB_redesign/beehive_PCB.png) should have two connectors for 2 independent 12V5A lines. Thus, it is possible to power more electronic devices rather than just using 1X 4pin conn. But again, it depends on how the board is _actually_ being used so if 1x 4pinn is more convenient so be it

amchagas, 2021-06-28:
Oh sorry I probably didn't express myself properly.. the connectors I am referring to are the digital signals that control the turning on/off of the rest of the circuit.. so they are normally low power 3.3v to 5v Volts

isobianin, 2021-06-28:
I see, no worries! I have just got a little confused over this