BeeHive-org/BeeHive #33: board suggestion: logic level shifter

amchagas, 2021-06-28:
the ESP32 runs on 3.3V logic levels, which is fine, since a lot of electronic circuits and devices are compatible with this.

However to make the system more user friendly, and more compatible with lab devices, we should adopt a board/circuit that makes logic level shifting automatic, with initial consideration being to shift between 3.3V to 5V and vice versa.

This has been implemented by several projects before and should not be complicated. there are even little boards being sold on Ebay/amazon that takes care of this, so we could simply use those? more details on the boards [here](https://www.ebay.co.uk/itm/272249807035?hash=item3f635a60bb:g:f1UAAOSwajFezLJl) They basically have 4 low level and 4 high level input/output that are connected via a circuit containing a mosfet. Then the user needs to supply a referece high voltage and a referece low voltage. I have used one of those to connect Beehive to other lab systems (running at 5V). For a high reference, I simply connected Beehive's 5V vcc and for the low reference, I got 3.3V with a voltage divider (maybe this is not the best approach though?)

In the case we want to make a board of our own, I would make sure the board manages to take more than 2 or 4 inputs, so that we keep the number of boards/cables down.

isobianin, 2021-06-28:
Hi Andre,

is this something similar to what you have posted and what you have in mind? https://www.sparkfun.com/products/12009
Seems like a perfect fit and with a very precise term of "bi-directional" level converter. I will look into it and see whether I can manage to do something similar or just employ a working design

amchagas, 2021-06-29:
Yes this is what I had in mind, which is quite similar to the ebay board posted above!

isobianin, 2021-07-02:
Hi Andre,

I would like to add some clarification to this issue:

1) How exactly did you hook Beehive up to the aforementioned converter? How did you connect 3.3V and 5V to the board?

2) Regarding design, how input and output connectors should look like? For the low side, 3.3V, I think we can use our standard 4pin connectors. But what do we use for the high side, 5V?

3) I thought about using through-hole transistors as usual, is that right?

amchagas, 2021-07-05:
Hi Ihor, 

1 - I just soldered things directly to a protoboard... on the protoboard I had 5V from Beehive as the high side VCC and for the 3.3V (Low Side VCC) I had a small voltage divider using resistors and the 5V line from beehive.

2 - I think the high side connector will depend on how many connectors will be coming from beehive? I mean, if we are using only one 4 pin connector, than each board should be able to do the level shifiting of 2 digital lines. In which case we could simply have 2 X 2pin connectors on the high side? This way one would have D1, GND, D2, GND (which would allow for digital signals to come from different devices). I think the same could be applied if we decide to go for more lines per board (which I think we should, if we can benefit from having more lines and a footprint that is similar to the other boards?)

3 - through-hole transistors are fine!

isobianin, 2021-07-08:
Hi Andre,

I have been looking into [this MOSFET](https://uk.rs-online.com/web/p/mosfets/6714736/?cm_mmc=UK-PLA-DS3A-_-google-_-CSS_UK_EN_Semiconductors_Whoop-_-MOSFETs_Whoop-_-6714736&matchtype=&aud-827186183686:pla-338650602187&gclid=Cj0KCQjwxJqHBhC4ARIsAChq4auKvChHT6tmbiaAmfcCc7USbWqSjcuLErJWPhumDCEDpTYisF-rgbIaAiOSEALw_wcB&gclsrc=aw.ds) and[ this schematic ](https://www.hobbytronics.co.uk/mosfet-voltage-level-converter)based of it. What do you think if we take this MOSFET? I chose it because it is, firstly, available on rs and ,secondly, because MOSFETs suitable for level shifting are tricky to get in through-hole package. I think the price is comparable to other MOSFETs we are using, isn't it?

Regarding how 5V are going to be converted into 3.3V, I think a simple voltage divider is really quite good solution, as you mentioned in the original post. I also was thinking about diodes, Zener diode and other stuff but all those things have their own tricks. Besides, I really wish to make it so that we have as few different components as possible so all components can be sourced relatively easy

isobianin, 2021-07-13:
Hi Andre,

please, have a look at the new [level-shifter board](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/level_shifter).
[image]
As I said before, it is built on 4 BS170 MOSFETs, has 4 low 3.3V channels and 4 high 5V channels. 3.3V are obtained via voltage divider.

amchagas, 2021-07-19:
thanks Ihor, this looks great!

closed 2021-07-19 by amchagas