BeeHive-org/BeeHive #35: 8 switch array voltage

isobianin, 2021-07-24:
Hi Andre,

it is related to [this issue](https://github.com/BeeHive-org/5-choice-serial-reaction-time/issues/1) from the rodent behaviour.

So the 4 switch array should work fine but I think 8 switch array has a little design overlook inside so I just want to double check on that with you so we are on the same page. Please, have a look at the schematics. The shift register is supplied with 5V and thus outputs 5V too.
[image]
The reason why it got my attention is that 4 switch array completely mimics what was done back in FlyPi where pins where directly connected to bases of transistors through current limiting resistors. The difference is that 4 switch array connects 3.3V to the base while 8 switch array connects 5v, or at least that is what I think is going on here. 

I do not think that it is right. Maybe transistors work fine but probably adding simple voltage divider will balance things out as 3.3V is more appropriate. I have checked the register's datasheet and see that it can take as low as 2V so we should be alright.

amchagas, 2021-07-27:
good catch! 

Maybe this won't be a problem for the transistor themselves, do I remember correctly that we are using PN2222 transistors in this design? If so, they are the same from the FlyPi and so the Arduino there, sends out 5V on its DO ports... All this at least from a electronics working point of view... But now that I write, you were probably already aware of that?! :see_no_evil: 

Now, do you mean that outputting 5V is not correct because the rest of the system is 3.3V?

isobianin, 2021-07-27:
Hi, thanks for clarification. To be honest, I got a bit confused in this one. So we were using 5V in FlyPi and now we are working with 3.3V. From what I see, it is not a problem here as this switch will work as intended. On the other hand, I think 4 switch array is not. I remember that I took the shifter out and just connected pins directly to base resistors so probably some alternations there will be needed

amchagas, 2021-07-27:
Now I am a bit confused :D .. I thought we had phased out the 4 switch array as it was not necessary, since we had the 8 switch one? As in, it is the same IC, so if one needs less pins, just leave some of the footprints in the board unpopulated?

isobianin, 2021-07-27:
That makes sense, though, we never actually talked about it :sweat_smile: . I thought that 4 switch array avoids paying more for the IC if someone just needs a basic switch. If we settle with the doctrine that 8 switch array is our only switch board, I think I am okay with that. We just need to evaluate this

amchagas, 2021-08-17:
do I have that wrong again, that the 4 switch array and the 8 switch array use the same IC? 
If they use the same, then the cost of buildiing an 8 X 4 switch array is not so big... if they are indeed different, what is the cost difference between them? Does it pay off for the added number of PCBs and parts list?

isobianin, 2021-08-18:
Hi Andre,
to quote our wiki:

[image]

What happens is that 4 switch array does not have the IC at all thus lowering the cost. If I am not mistaken, the shift register costs **about 1 dollar**, maybe a bit less which is relatively high. The idea was to omit it and use high/low signals from the ESP32 itself to do the work. To make it work, we need to change resistors we have now to ones with lower resistance.

On the other hand, I think 8 switch array is indeed more versatile and it is also much better protected as the level shifter also is responsible for handling any voltage spike and other shenanigans. 

I hope it answers your questions

amchagas, 2021-08-20:
it does, thanks! I still have the older 4 switch arrays with shift registers here... I think this is what got me confused!

closed 2023-03-16 by amchagas