BeeHive-org/BeeHive #36: Grove-To-BeeHive adaptor

isobianin, 2021-08-13:
BeeHive is compatible with Grove because of its data protocols used and connector pinouts. The latter, though, has an implication - connectors are not identical. Therefore, we need to design a certain way of making these systems mechanically compatible. With this in mind, the following could be done:
- a small PCB with BeeHive connector on one side and Grove connector on the other
- an addition to the abovementioned point is to make a relatively big break-away board so one can order it and have a lot of adaptors straight away while also lowering the cost of an each adaptor
- a simple adaptor board with a lot of BeeHive connectors and Grove connectors

I believe that the first two variants are rather good.

isobianin, 2021-08-13:
Just a quick update, https://github.com/BeeHive-org/BeeHive/tree/master/hardware/grove_adaptor
The connector that grove uses seems to be [proprietary](https://arduino.stackexchange.com/questions/9030/what-type-of-connector-does-the-grove-system-use). [Here](https://www.seeedstudio.com/Grove-Universal-4-pin-connector.html) are the connectors needed for the board. PCB footprint is taken from [here](https://www.seeedstudio.com/Grove-Gas-Sensor-MQ9.html). I placed BeeHive and Grove logos on the respective side of the board so it is easier to distinguish though they differ significantly by their size.

isobianin, 2021-08-14:
Added a panelised version as promised. Here, how it looks:
[image]

amchagas, 2021-08-17:
That looks great! 

Interesting to learn that they might be using a proprietary connector... I took another look at their wiki and indeed could not find more details (other than drawings and schematics) like the manufacturer part number (I might be mistaken to assume that Seeed does not make them). But I found this part of the wiki: https://wiki.seeedstudio.com/Grove_System/#grove-connector

and the link to buy the connectors: https://www.seeedstudio.com/Grove-Female-Header-SMD-4P-2.0mm-20Pcs-p-4589.html

isobianin, 2021-08-18:
Hi Andre,
it is a connector that someone manufactures specifically for them. It does not align with anything that is out there on the market, it is clearly proprietary which is strange for the open-source project. Indeed, its documentation is obfuscated for some reason.
Just a few more links on this one:
https://cpc.farnell.com/seeed-studio/110990030/grove-connector-4p-2mm-10pk/dp/MK00368
https://www.amazon.co.uk/Seeedstudio-Grove-Universal-Connector-Connectors/dp/B01AD632S2

amchagas, 2021-08-20:
cool, thanks!