BeeHive-org/BeeHive #38: all boards review

amchagas, 2022-01-15:
as we use this system more and more, we would like to review all the boards. This comes to improve things, adapt initial development that took a different direction, etc.

Main board suggestions:
- [ ]  get rid of the 5V plane and make it into a trace
- [ ] remove the USB 5V supply and the second power conversion module

New board suggestion:
- [ ] create a 12 to 5V board that has a USB port as output (or adapt existing ones)

isobianin, 2022-01-18:
- [ ] [level shifter](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/level_shifter) might need a Zener diode to properly convert 5V to 3.3V as of right now the output can vary depending on the load that is connected to 3.3V. On the other hand, if the level shifter is solely used with low loads aka ports of microcontrollers, it will work fine even as is

isobianin, 2022-01-18:
- [ ] [IR sensors array](https://github.com/BeeHive-org/BeeHive/tree/master/hardware/ir_sensor) absolutely needs at the very least a Zener diode in 5V to 3.3V line to function properly

isobianin, 2022-01-18:
So I think it would be nice to get some testing, I guess? If things work the way it is - I am fine with that. I do not want to overcomplicated something that works. The problem with voltage divider may arise in the following cases:
1) IR board is fully loaded with phototransistors. If every connection is used, it might happen that the voltage divider may not supply the given voltage because of the resistance that changed because of all the load. On the other hand, 5V that is being converted to the voltage divider in the first place is regulated, so it might work fine
2) Level shifter board should not have problems as is because currents are low. But can it be that there are problems when working on high frequencies?

I honestly think that in the level shifter scenario we can even use one of the pins as a voltage supply and omit voltage divider at all

isobianin, 2022-01-18:
Possible solution to 5V to 3.3V DC issue is https://www.mouser.co.uk/ProductDetail/STMicroelectronics/LD1117V33?qs=sGAEpiMZZMug9GoBKXZ75%252BNB3YkcECi9WejQbqCGDNE%3D