BeeHive-org/BeeHive #51: Grove adapter panel has small gaps in outline

kasbah, 2022-12-30:
This is causing issues with the preview rendering [on kitspace.org](https://kitspace.org/boards/github.com/beehive-org/beehive/grove_adaptor_panel/). One day our renderer hopefully will be able to fix cope with these but it will be a while. 

https://github.com/kitspace/kitspace-v2/issues/534 

[image]

amchagas, 2023-03-16:
Hi Kaspar,

thanks for flagging this. We looked a bit more into it, and there seems to be something wrong with the grid in Kicad, not sure if it is due to incompatibility between Kicad versions, or between Kicad being deployed in different OSs (windowsXlinux) since once we open the editor and change the cutedge lines, they snap into grid points in a way we can never get them to form straight lines (unless we change both sides of all lines). Is this something you've seem before and might have a tip/hint on how to solve it?

kasbah, 2023-03-17:
You should be able to set the "User Grid" to a really small size (that effectively means there's no grid constraint). I haven't worked with it in a long time so sorry I can't be of more help.